RobotLocomotion/pytorch-dense-correspondence #154: How to get the data?

Wegnery219, 2018-07-31:
I don't understand where the data come from.What is spartan?If I want to evaluate the network,do I need to train first?

manuelli, 2018-08-02:
Yes we will do this eventually, along with some sample data so you can evaluate the network.

Wegnery219, 2018-08-03:
Thank you for your reply.

closed 2018-08-03 by Wegnery219

peteflorence, 2018-09-06:
Hi @Wegnery219, @jiamingsuen --
We have now made all of our data available!
This [tutorial](https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/master/doc/tutorial_getting_started.md) will walk through downloading the data and training a model.
If you are familiar with this type of software and have a new-ish GPU, I estimate only would take ~30 minutes to get a model trained and results visualized.
Let me know if you get caught up, or we could make anything more clear?