RobotLocomotion/pytorch-dense-correspondence #157: Why can't I import the python files in other folders?

nemocandy5, 2018-08-14:
Excuse me. I tried to run the notebook dense_correspondence/dataset/simple_datasets_test.ipynb
But it has import error. As shown below. If 22 lines are annotated, there are still similar problems in line 23 & line 24. How can I solve this?

[image]

manuelli, 2018-08-14:
This is because you have not added the module onto your `PYTHONPATH`. Make sure you run `use_pytorch_dense_correspondence` and `use_director`. 

Before posting issues note that the tutorials are still under construction, and please check the readme's we have for setup. They explicitly mention the step about sourcing environment variables.

closed 2018-08-14 by manuelli

peteflorence, 2018-09-06:
Hi @nemocandy5 -- just looking at this now, were you able to get this to work, or need any more help?  We also just posted a [tutorial](https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/master/doc/tutorial_getting_started.md) which might help you out.

Looking at the error message -- this may be the key command lines you need while setting up?

```
git submodule update --init --remote --recursive
git submodule update --recursive
```

Let me know if you have any more questions / issues.

peteflorence, 2018-09-07:
And if not then yes, running `use_pytorch_dense_correspondence` in the terminal session will set your PYTHONPATH to be happy.  Should do this before starting a jupyter notebook server.