RobotLocomotion/pytorch-dense-correspondence #171: On Ubuntu 16.04 docker_build.py finish with 403: Forbidden

iskanderuse, 2018-10-09:
On Ubuntu 16.04 docker_build.py finish with that log:
+ build_director
+ director_version=0.1.0-266-g071a233
+ director_binary=ubuntu-16.04
+ wget https://dl.bintray.com/patmarion/director/director-0.1.0-266-g071a233-ubuntu-16.04.tar.gz
--2018-10-09 17:12:26--  https://dl.bintray.com/patmarion/director/director-0.1.0-266-g071a233-ubuntu-16.04.tar.gz
Resolving dl.bintray.com (dl.bintray.com)... 159.122.18.156
Connecting to dl.bintray.com (dl.bintray.com)|159.122.18.156|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2018-10-09 17:12:27 ERROR 403: Forbidden.

Is it proprietary too?

peteflorence, 2018-10-09:
Hey @iskanderuse, this is not proprietary... potentially the dl.bintary.com server is down?  I can look into it

peteflorence, 2018-10-09:
Hi there, this download server is also down for me, I just confirmed.  You don't need that piece of software, (Director), just to run through for example our tutorial.  It was used to generate part of the data we provided already.

You can temporarily disable its installation by just commenting out this line: https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/master/docker/pytorch-dense-correspondence.dockerfile#L33 with a `#`

iskanderuse, 2018-10-09:
Ok, Thank You!

manuelli, 2018-10-11:
Confirmed the server is down, we are working on a fix.

peteflorence, 2018-10-12:
#176 is a temporary fix.  Let's keep this issue open though until we resolve the director binaries.

peteflorence, 2018-10-12:
I changed the name of this issue, since docker_build.py will finish now on master

We just need to fix the availability of director binaries, for parts of the codebase that use director.  (ie background subtraction)

manuelli, 2018-11-04:
The bintray server is back up.

closed 2018-11-04 by manuelli