RobotLocomotion/pytorch-dense-correspondence #173: ImportError: cannot load CLoader

domantasjurkus, 2018-10-11:
Stumbled upon this issue when running the tutorial notebook. In case anyone's stuck, follow [these steps](https://github.com/yaml/pyyaml/issues/108) inside the container. In short:

1. `sudo apt install libyaml-dev`
2. `sudo pip --no-cache-dir install --verbose --force-reinstall -I pyyaml`

The container root password is `password`

peteflorence, 2018-10-11:
Thanks for posting your solution. What was the issue that you encountered
with yaml?

manuelli, 2018-10-11:
It seems like yaml was installed in the [install_director.sh](https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/master/docker/install_director.sh) script. If you commented that out in the dockerfile (as was suggested in #171 ) then yaml wouldn't be installed. I created a patch #174 that just adds yaml to our standard list of dependencies.

peteflorence, 2018-10-12:
This issue should be fixed by #176

closed 2018-10-12 by peteflorence