RobotLocomotion/pytorch-dense-correspondence #175: ImportError: libSM.so.6: cannot open shared object file: No such file or directory

markovcat, 2018-10-12:
In the training tutorial notebook, stuck on the first cell:
```
ImportError                               Traceback (most recent call last)
<ipython-input-1-be2ab2db5d25> in <module>()
      1 import dense_correspondence_manipulation.utils.utils as utils
      2 utils.add_dense_correspondence_to_python_path()
----> 3 from dense_correspondence.training.training import *
      4 import sys
      5 import logging

/home/josh/code/dense_correspondence/training/training.py in <module>()
     41 from dense_correspondence.loss_functions.pixelwise_contrastive_loss import PixelwiseContrastiveLoss
     42 import dense_correspondence.loss_functions.loss_composer as loss_composer
---> 43 from dense_correspondence.evaluation.evaluation import DenseCorrespondenceEvaluation
     44 
     45 

/home/josh/code/dense_correspondence/evaluation/evaluation.py in <module>()
      7 utils.add_dense_correspondence_to_python_path()
      8 import matplotlib.pyplot as plt
----> 9 import cv2
     10 import numpy as np
     11 import pandas as pd

/usr/local/lib/python2.7/dist-packages/cv2/__init__.py in <module>()
      1 import importlib
      2 
----> 3 from .cv2 import *
      4 from .data import *
      5 
```
I encountered the other issues and followed your suggestions in #174 .

manuelli, 2018-10-12:
Hey @markovcat,

Did you disable the director build as specified in #171? Our server holding the binaries is down and is wreaking havoc on our system. We are working on getting a new server up to hold the binaries, at which point everything should be fixed. Hopefully that will be done soon. Otherwise you can just try commenting out [this](https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/master/docker/install_director.sh#L41) one line. That disables the director build, but still installs it's dependencies.

peteflorence, 2018-10-12:
Good call to instead comment out that line!

peteflorence, 2018-10-12:
Merged into master: #176

closed 2018-10-12 by peteflorence