RobotLocomotion/pytorch-dense-correspondence #177: Change Detection Pipeline can not find directorPython

iskanderuse, 2018-10-31:
Script run_change_detection_pipeline.py throws error: directorPython not found inside pdc.
What is the "directorPython"? And how to find it?

iskanderuse, 2018-10-31:
I try to I'm trying to apply pdc on my robot installation. Director binaries are available now. May be this issue due to commented #build_director in master branch.. (https://github.com/RobotLocomotion/pytorch-dense-correspondence/issues/171#issuecomment-428281322)

manuelli, 2018-10-31:
Yes you are exactly right. Now that the director binaries are available you should be able to build director as part of the docker build. @iskanderuse if you know what to comment/uncomment that's worth a try. We will submit a PR with a fix as soon as we can.

manuelli, 2018-11-04:
Fixed by #179

closed 2018-11-04 by manuelli