RobotLocomotion/pytorch-dense-correspondence #181: Can I have more specific details for 3D change detection ?

zy871125746, 2018-11-08:
Hello everyone !
I am trying to reimplement your triainning procedure with my code. Now I am working on change detection part, can i have more details about this algrithm ?

manuelli, 2018-11-08:
Hi,

As mentioned in our paper we use the approach from the following paper

> R. Finman, T. Whelan, M. Kaess, and J. J. Leonard. Toward lifelong object segmentation from
> change detection in dense rgb-d maps. In Mobile Robots (ECMR), 2013 European Conference on,
> pages 178–185. IEEE, 2013
> 

If you have additional knowledge (i.e. that your objects are on a table for example) then you could also just crop your 3D reconstruction. Any approach that works for removing the background will be fine.

closed 2018-11-08 by manuelli

zy871125746, 2018-11-09:
Yes, I have read the papar that you have mentioned. In that paper, I am a little confuse about about that free space filtering so I am trying use my own way to remove the background now.

Thank you for your helping.