RobotLocomotion/pytorch-dense-correspondence #183: What is blind_non_match_loss in training.py ?

zy871125746, 2019-01-05:
Could you explain what is the blind_non_match_loss ? I can not find this loss in the paper

peteflorence, 2019-01-07:
Hi there —

The non match sampling that we are doing is the same as what we understood
is described in Schmidt et al 2017, which is to sample non matches in image
B for the pixels in image A which have a match in image B.

One thing we investigated was whether there was an issue with this since
opposite sides of the object wouldn’t end up sampled as being non matches.
If you think about it, if two views are looking at opposite views of the
object, then there won’t be matches for many pixels, and so those pixels
won’t end up having any non matches sampled.

The “blind non matches” are accordingly non match samples in image B for
even the pixels in image A for which there isn’t a match.

After experimentation though we didn’t observe any significant differences
in quantitative metrics when using blind non match sampling. And yes this
isn’t mentioned in the paper — it’s a pretty minor note compared to other
experiments mentioned.

Does that make sense?

zy871125746, 2019-01-15:
Sorry for answering later. After your explantion, I figured out how the "blind non matches" works. Thanks a lot !

closed 2019-01-22 by peteflorence