RobotLocomotion/pytorch-dense-correspondence #184: python2 incompatibility in `setup.py` (visdom==0.1.8.7)

KuKuXia, 2019-01-15:
Problem:
OS: Ubuntu 16.04
When I run the command ./docker_build.py followed by the installation guide. The errors occurred in
install_pytorch.sh file:

```bash
Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-jqLhlI/visdom/setup.py", line 35, in <module>
        readme = open('README.md', 'rt', encoding='utf8').read()
    TypeError: 'encoding' is an invalid keyword argument for this function
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-jqLhlI/visdom/
```
which is caused by the visdom package. There is a problem in the newer version: 0.1.8.7 which has been discussed in [Visdom Issues: #548](https://github.com/facebookresearch/visdom/issues/548). 
My solution is:
Change the line in install_pytorch.sh files:
from:
```bash
sudo pip install visdom
```
to:
```bash
sudo pip install visdom==sudo pip install visdom==0.1.8.6
```
So everything was ok and I finished the installation guide successfully. Hope this could help others.

peteflorence, 2019-01-22:
Glad you found a way around it, and thanks for posting your fix!

My personal recommendation would be to just use tensorboard rather than visdom, I think will PR a change that does this.

Thanks!

manuelli, 2019-05-27:
This should be fixed once the latest PR #196 lands.

manuelli, 2019-06-26:
Should be fixed in #199

closed 2019-06-26 by manuelli