RobotLocomotion/pytorch-dense-correspondence #185: [Question] Is robot controller's source code available ?

gachiemchiep, 2019-01-29:
Hello sir.
Thank you for your great work.

Is the source code for robot controller is public too?
If it is, could you share me the link.

peteflorence, 2019-01-30:
Hi, thanks for the interest in our work.  Yes this repo holds just the vision (dense descriptors) part of the code.  Yes most of the robot code is also available, except for a couple private submodules, it is public in this repo: https://github.com/RobotLocomotion/spartan

Note however we don't really document too much of that code / intend it to be for public use.  But if you poke around enough, you might learn something, and/or find code that would be useful to you.  It is all BSD 3-clause.

Good luck with your work!

closed 2019-01-30 by peteflorence

gachiemchiep, 2019-01-30:
@peteflorence 
Thank you. i will take a look.