RobotLocomotion/pytorch-dense-correspondence #186: F tensorflow/core/platform/cpu_feature_guard.cc:37] The TensorFlow library was compiled to use AVX instructions, but these aren't available on your machine.

tpatten, 2019-01-30:
I am running Step 7 (training a network) and get this error in the terminal when executing the notebook `dense_correspondence/dataset/simple_datasets_test.ipynb`. When the error occurs, the notebook itself crashes with the error
```
The kernel appears to have died. It will restart automatically.
```
I am running with a GTX 1070Ti.

peteflorence, 2019-01-30:
Hi @tpatten, hm yes this an odd set of errors, we have not seen this before.

It's especially odd because we are not using TensorFlow.

Are you using the docker as suggested?

Another piece of info, can you just do this in an otherwise empty notebook?

```python
import torch
A = torch.rand((4096,4096)).cuda()
C = A*A
print A
```

tpatten, 2019-01-30:
Thanks @peteflorence for the quick response. I agree that these are very strange errors.

I am using the docker as suggested in [here](https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/master/doc/tutorial_getting_started.md), with a fresh clone of the repository.

The simple test (Step 6) worked fine and in fact, Step 7 worked on my personal machine, albeit not enough GPU memory to finish. So this error seems specific to the particular machine that I am testing with.

The snippet works fine with output:
```
 9.8629e-01  4.2406e-01  1.1165e-01  ...   2.9767e-01  5.8334e-01  1.2605e-02
 3.8082e-02  9.1599e-01  8.5884e-01  ...   4.1615e-01  9.6462e-01  3.9678e-01
 3.8218e-01  3.6932e-01  2.8543e-01  ...   8.7817e-01  1.3873e-01  9.0384e-01
                ...                   ⋱                   ...                
 9.9120e-02  1.3413e-01  4.6128e-01  ...   7.8200e-01  3.5384e-01  7.0051e-02
 1.1918e-01  3.4889e-01  8.5294e-01  ...   5.0268e-01  4.7582e-01  9.4590e-01
 9.7040e-01  8.0176e-01  4.8712e-01  ...   5.3505e-01  4.1472e-01  5.4238e-01
[torch.cuda.FloatTensor of size 4096x4096 (GPU 0)]
```

peteflorence, 2019-01-30:
OK one thing to consider would be to remove this line and Line 19: https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/d71be2a129e413859611ede6fb613563160c5f5d/docker/install_more.sh#L18
Because although we don't use tensorflow, it is installed in the docker image.

If you remove that line, then tensorflow won't even be present in the docker image.  You may need to remove any calls however to `tensorboard` (which we do use).  Easiest way to do this would be to just comment out any lines that give you errors when trying to run scripts that use tensorboard.

This will be informative because either:
1. if the error goes away, then somehow the installed tensorflow is messing stuff up, and we can go fix it
2. if the error doesn't go away, then something about your docker image setup isn't right (because tensorflow shouldnt be in there)

tpatten, 2019-01-31:
Ok thanks. I will try that and let you know. Out of curiosity, since you don't use tensorflow how come it is installed in the docker image, e.g. https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/d71be2a129e413859611ede6fb613563160c5f5d/docker/install_more.sh#L18 and https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/d71be2a129e413859611ede6fb613563160c5f5d/docker/install_dependencies.sh#L32. My guess is that there are some other dependencies that are needed.

peteflorence, 2019-01-31:
Yes suggest removing both of those lines and then rebuild the image. I believe the answer to your question of why it’s there is just that at some point we were trying out something in tensorflow in here. Shouldn’t be needed now though.

tpatten, 2019-01-31:
It was sufficient to remove the commands that install `tensorflow` and `tensorboard`. I left `tensorboard_logger` in there and it works.

peteflorence, 2019-01-31:
Interesting. Did the error go away?

tpatten, 2019-01-31:
Yes completely

peteflorence, 2019-01-31:
Hm. Glad it went away.

What CPU are using? Looking again at the error, it’s conplaining about AVX,
which is a CPU thing. Trying to understand if this error will affect other
users

tpatten, 2019-01-31:
I also believe it is a CPU specific problem as on my other machine I did not have this issue. For your reference, the output of `lscpu`

```
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    1
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 30
Model name:            Intel(R) Core(TM) i5 CPU         760  @ 2.80GHz
Stepping:              5
CPU MHz:               1199.000
CPU max MHz:           2800,0000
CPU min MHz:           1199,0000
BogoMIPS:              5600.27
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              8192K
NUMA node0 CPU(s):     0-3
```

closed 2019-02-15 by tpatten