RobotLocomotion/pytorch-dense-correspondence #190: docker configuration

BenjaminLAZARD, 2019-04-09:
Hello,

When running `sudo ./docker_build.py`

apt-get seem to be unable to access channels from the script.
I already configured docker so that it could go through a proxy, and `docker run hello-world` runs without problems.

```bash
[sudo] password for benj: 
building docker container . . . 
building docker image named  root-pytorch-dense-correspondence
command = 
 
 docker build --build-arg USER_NAME=root                        --build-arg USER_PASSWORD=password                      --build-arg USER_ID=0                   --build-arg USER_GID=0 -t root-pytorch-dense-correspondence -f pytorch-dense-correspondence.dockerfile .

executing shell command
Sending build context to Docker daemon  17.92kB
Step 1/25 : FROM nvidia/cuda:8.0-devel-ubuntu16.04
 ---> ed7c1b380cab
Step 2/25 : ARG USER_NAME
 ---> Using cache
 ---> f6a706272afb
Step 3/25 : ARG USER_PASSWORD
 ---> Using cache
 ---> ddd86243972d
Step 4/25 : ARG USER_ID
 ---> Using cache
 ---> 594b23ecd76e
Step 5/25 : ARG USER_GID
 ---> Using cache
 ---> 6f1c470e7d07
Step 6/25 : RUN usermod -aG sudo $USER_NAME
 ---> Using cache
 ---> cbf2c606477e
Step 7/25 : RUN yes $USER_PASSWORD | passwd $USER_NAME
 ---> Using cache
 ---> e29049757f77
Step 8/25 : RUN usermod -u $USER_ID $USER_NAME
 ---> Using cache
 ---> 2238cff7e729
Step 9/25 : RUN groupmod -g $USER_GID $USER_NAME
 ---> Using cache
 ---> 966724663c3f
Step 10/25 : WORKDIR /home/$USER_NAME
 ---> Using cache
 ---> ff859bd0f803
Step 11/25 : COPY ./install_dependencies.sh /tmp/install_dependencies.sh
 ---> 4af4edfbde1a
Step 12/25 : RUN yes "Y" | /tmp/install_dependencies.sh
 ---> Running in 53bc243c64b8
+ apt-get update
Err:1 http://archive.ubuntu.com/ubuntu xenial InRelease
  Temporary failure resolving 'archive.ubuntu.com'
Err:2 http://security.ubuntu.com/ubuntu xenial-security InRelease
  Temporary failure resolving 'security.ubuntu.com'
Err:3 https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1604/x86_64  InRelease
  Could not resolve host: developer.download.nvidia.com
Err:4 http://archive.ubuntu.com/ubuntu xenial-updates InRelease
  Temporary failure resolving 'archive.ubuntu.com'
Err:5 http://archive.ubuntu.com/ubuntu xenial-backports InRelease
  Temporary failure resolving 'archive.ubuntu.com'
Reading package lists...
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial/InRelease  Temporary failure resolving 'archive.ubuntu.com'
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial-updates/InRelease  Temporary failure resolving 'archive.ubuntu.com'
W: Failed to fetch http://archive.ubuntu.com/ubuntu/dists/xenial-backports/InRelease  Temporary failure resolving 'archive.ubuntu.com'
W: Failed to fetch http://security.ubuntu.com/ubuntu/dists/xenial-security/InRelease  Temporary failure resolving 'security.ubuntu.com'
W: Failed to fetch https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1604/x86_64/InRelease  Could not resolve host: developer.download.nvidia.com
W: Some index files failed to download. They have been ignored, or old ones used instead.
+ apt install --no-install-recommends terminator tmux vim gedit git openssh-client unzip htop libopenni-dev apt-utils usbutils dialog python-pip python-dev ffmpeg

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
Package dialog is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package apt-utils is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package terminator
E: Unable to locate package tmux
E: Unable to locate package vim
E: Unable to locate package gedit
E: Unable to locate package git
E: Unable to locate package openssh-client
E: Unable to locate package unzip
E: Unable to locate package htop
E: Unable to locate package libopenni-dev
E: Package 'apt-utils' has no installation candidate
E: Unable to locate package usbutils
E: Package 'dialog' has no installation candidate
E: Unable to locate package python-pip
E: Unable to locate package python-dev
E: Unable to locate package ffmpeg
The command '/bin/sh -c yes "Y" | /tmp/install_dependencies.sh' returned a non-zero code: 100
```

manuelli, 2019-04-09:
You shouldn't run docker_build.py with sudo. I think that is the problem. The readme contains instructions on how to run docker without sudo priveleges

BenjaminLAZARD, 2019-04-09:
Thank you for your quick reply. Of course I did start by following exactly the instructions of the readme. The only difference is the `sudo`. Without it, I get the following error.

```bash
building docker container . . . 
building docker image named  benj-pytorch-dense-correspondence
command = 
 
 docker build --build-arg USER_NAME=benj                        --build-arg USER_PASSWORD=password                      --build-arg USER_ID=1000                        --build-arg USER_GID=1000 -t benj-pytorch-dense-correspondence -f pytorch-dense-correspondence.dockerfile .

executing shell command
Sending build context to Docker daemon  17.92kB
Step 1/25 : FROM nvidia/cuda:8.0-devel-ubuntu16.04
 ---> ed7c1b380cab
Step 2/25 : ARG USER_NAME
 ---> Using cache
 ---> f6a706272afb
Step 3/25 : ARG USER_PASSWORD
 ---> Using cache
 ---> ddd86243972d
Step 4/25 : ARG USER_ID
 ---> Using cache
 ---> 594b23ecd76e
Step 5/25 : ARG USER_GID
 ---> Using cache
 ---> 6f1c470e7d07
Step 6/25 : RUN usermod -aG sudo $USER_NAME
 ---> Running in f3f8045d4d3f
usermod: user 'benj' does not exist
The command '/bin/sh -c usermod -aG sudo $USER_NAME' returned a non-zero code: 6
```

peteflorence, 2019-04-09:
Hi Benj --
What if you run these following commands in the terminal? First run `whoami` and see what you get.  Then open a python interpreter and call `getpass.getuser()`.  Here's the example for me below.  What do you get?

```
peteflo@peteflo-m2:~$ whoami
peteflo
peteflo@peteflo-m2:~$ python
Python 2.7.15rc1 (default, Nov 12 2018, 14:31:15) 
[GCC 7.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import getpass
>>> getpass.getuser()
'peteflo'
```
Also for context we've used this docker build configuration on dozens on different computers with lots of different configurations.  I imagine we can get yours to work just fine.

BenjaminLAZARD, 2019-04-10:
Thank you!
Both commands return my username, "benj".

manuelli, 2019-05-27:
I have run through the entire tutorial in #196 and didn't encounter this issue. You may want to try again once that PR lands and see if you still have problems.

closed 2019-05-27 by manuelli