RobotLocomotion/pytorch-dense-correspondence #191: About ply file in PDC

t0mo3, 2019-04-12:
Nice to meet you. 

We are now interested in pytorch-dense-correspondence technology and try it out in our own developing environment.
However, rendering after 3D construction is not doing well and we have faced in a problem.
If possible, could you advise us on  question below?

Question：Is it impossible to read the ply file that we created? 
                  How can we solve this problem?
Symptom：”vtkArray = dataObj.GetPoints().GetData()” occurs in “run_change_detection.py”.
Cause：The header information of our ply file is different from the example.
How to create our ply file：refer to https://github.com/andyzeng/tsdf-fusion-python

➢example’s header
ply..
format.binary_little_endian.1.0..
element.vertex.57294..
property.float.x..
property.float.y..
property.float.z..
element.face.113352..
property.list.uchar.int.vertex_indices..
end_header.

➢ our’s
ply
format binary_little_endian 1.0
comment VCGLIB generated
element vertex 15716
property float x
property float y
property float z
property uchar red
property uchar green
property uchar blue
property uchar alpha
element face 31316
property list uchar int vertex_indices
end_header

➢ error log
/home/user/director/lib/python2.7/site-packages/vtk/util/numpy_support.py:134: FutureWarning: Conversion of the second argument of issubdtype from `complex` to `np.complexfloating` is deprecated. In future, it will be treated as `np.complex128 == np.dtype(complex).type`.
  assert not numpy.issubdtype(z.dtype, complex), \
Traceback (most recent call last):
  File "./modules/dense_correspondence_manipulation/scripts/run_change_detection.py", line 81, in <module>
    run(data_folder, config_file=args.config_file, debug=args.debug, globalsDict=globalsDict)
  File "./modules/dense_correspondence_manipulation/scripts/run_change_detection.py", line 41, in run
    background_data_folder=background_scene_data_folder)
  File "/home/user/code/modules/dense_correspondence_manipulation/change_detection/change_detection.py", line 495, in from_data_folder
    foreground_reconstruction = TSDFReconstruction.from_data_folder(data_folder, config=config)
  File "/home/user/code/modules/dense_correspondence_manipulation/fusion/fusion_reconstruction.py", line 352, in from_data_folder
    fr.setup()
  File "/home/user/code/modules/dense_correspondence_manipulation/fusion/fusion_reconstruction.py", line 288, in setup
    self.load_poly_data()
  File "/home/user/code/modules/dense_correspondence_manipulation/fusion/fusion_reconstruction.py", line 294, in load_poly_data
    self.crop_poly_data()
  File "/home/user/code/modules/dense_correspondence_manipulation/fusion/fusion_reconstruction.py", line 234, in crop_poly_data
    self.poly_data = director_utils.cropToBox(self.poly_data, transform, dimensions, data_type=self.poly_data_type)
  File "/home/user/code/modules/dense_correspondence_manipulation/utils/director_utils.py", line 171, in cropToBox
    polyData = cropToLineSegment(polyData, origin - cropAxis, origin + cropAxis)
  File "/home/user/code/modules/dense_correspondence_manipulation/utils/director_utils.py", line 152, in cropToLineSegment
    polyData = labelPointDistanceAlongAxis(polyData, axis, origin=point1, resultArrayName='dist_along_line')
  File "/home/user/director/lib/python2.7/site-packages/director/segmentationroutines.py", line 214, in labelPointDistanceAlongAxis
    points = vtkNumpy.getNumpyFromVtk(polyData, 'Points')
  File "/home/user/director/lib/python2.7/site-packages/director/vtkNumpy.py", line 44, in getNumpyFromVtk
    vtkArray = dataObj.GetPoints().GetData()
AttributeError: 'NoneType' object has no attribute 'GetData'

I appreciate your kind response.

manuelli, 2019-04-12:
So our codebase uses VTK to read ply files. So it seems like there is some sort of incompatibility between the ply file you generated and the type that VTK can read. I have encountered such issues with PCL's ply writier. But in general this is a VTK compatibilty issue rather than something specific to our repo so I think that googling around the internet is your best bet.

t0mo3, 2019-04-17:
Thank you for your comment. I understand that there are some incompatibility between the file and VTK can read. Keep on figuring out. Thanks.

closed 2019-05-06 by manuelli