RobotLocomotion/pytorch-dense-correspondence #193: invalid tensor shape

Ademord, 2019-05-13:
Hello!
How cold the correct shape (the additional [1,1]) be fixed?

[image]

priyasundaresan, 2019-05-26:
Having the same issue,^ did you manage to resolve it?

priyasundaresan, 2019-05-26:
Resolved. In train.py, make sure that line 153 is changed to:
self.fc = nn.Linear(512 * block.expansion, num_classes) instead of:
self.fc = nn.Conv2d(512 * block.expansion, num_classes, (1, 1))

If you do:
print(self.fc.weight.size()), you'll see the torch.Size is now (1000, 512)

kingsman142, 2019-05-27:
@priyasundaresan I was just having this problem. Your solution worked perfectly!

manuelli, 2019-05-27:
Thanks for the solution @priyasundaresan. I will look into this and PR a fix.

manuelli, 2019-05-27:
I have just tested the full tutorial against the branch in PR #196 and everything seems to be working fine. Note that the new PR requires `nvidia-docker2` instead of just `nvidia-docker` so you may need to re-install that if you are on an older version of `nvidia-docker`.

closed 2019-05-27 by manuelli

priyasundaresan, 2019-05-27:
Thanks for your quick response and help @manuelli!