RobotLocomotion/pytorch-dense-correspondence #194: Outdated PyTorch in Docker build

kingsman142, 2019-05-27:
I'm currently using an RTX 2080 GPU with CUDA 10.0. I kept on having issues where training the network wouldn't work since nothing would be loaded onto the GPU. Installing PyTorch locally for CUDA 10.0 did not work. So, I discovered I had to modify this script and change the pytorch installations for CUDA 10.0:

**pytorch-dense-correspondence/docker/install_pytorch.sh**

```
sudo pip install https://download.pytorch.org/whl/cu100/torch-1.1.0-cp27-cp27mu-linux_x86_64.whl
sudo pip install https://download.pytorch.org/whl/cu100/torchvision-0.3.0-cp27-cp27mu-linux_x86_64.whl

sudo pip3 install https://download.pytorch.org/whl/cu100/torch-1.1.0-cp35-cp35m-linux_x86_64.whl
sudo pip3 install https://download.pytorch.org/whl/cu100/torchvision-0.3.0-cp35-cp35m-linux_x86_64.whl
```

I think it would be best if there was a disclaimer to change these installation URLs in the URL to stay up-to-date with both PyTorch (currently listed as 0.3.0 in the install script) as well as CUDA. I can change create a pull request if you want.

manuelli, 2019-05-27:
Hi @kingsman142. Yes the pytorch version was pinned to 0.3.0 since that was what was originally used when computing the results for the paper. Updating to pytorch 1.0 is fairly easy (we have done it on a private branch) but we just haven't ported it over to the public repo yet. Hopefully I will open a PR in the next few days with the update. 

In the meantime feel free to make a PR with your updates. Good suggestion to add a disclaimer, probably the README is the best place to that.

manuelli, 2019-05-27:
There is now an open PR #196 that updates the repo to pytorch 1.1

manuelli, 2019-06-26:
#199 updates the repo to work with pytorch 1.1

closed 2019-06-26 by manuelli