RobotLocomotion/pytorch-dense-correspondence #197: problem  while running training.ipynb -help required

Amulya21, 2019-06-24:
Hii,
 I am facing a problem while training, it is working fine it 371st line of training.py (i found this using print statements) later  using pdb i found that it is going to the path (python2.7/dist-packages/torch/nn/functional.py )in this functional.py at line 90 return f(input,weight,bias) there is some issue i  guess.
Please help
Thanks

manuelli, 2019-06-26:
Hi,

Sorry you ran into issues. PR #199 just updates the repo to pytorch 1.1. I ran through the entire tutorial from a fresh clone and docker build and didn't encounter any issues. Note that you should run the notebook [`training_tutorial.ipynb`](https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/master/dense_correspondence/training/training_tutorial.ipynb), i'm not sure which file you meant by `training.ipynb`.

Hopefully with these updates if you follow the tutorial start again from the beginning you shouldn't encounter any issues. But please let us know if you do.

Amulya21, 2019-06-30:
Thanks for the reply.
I actually did run training_tutorial.ipynb itself but i was unable to train it that time but the problem got resolved with older version of your code itself and i was able train the model.

closed 2019-06-30 by manuelli