RobotLocomotion/pytorch-dense-correspondence #200: Problem with training_tutorial.ipynb

kevinkengne, 2019-07-01:
dense_correspondence/dataset/simple_datasets_test.ipynb runs fine. However, I get the following error when I run dense_correspondence/training/training_tutorial.ipynb:

TypeError                                 Traceback (most recent call last)
<ipython-input-3-77b56ac20ed4> in <module>()
      5     print "training descriptor of dimension %d" %(d)
      6     train = DenseCorrespondenceTraining(dataset=dataset, config=train_config)
----> 7     train.run()
      8     print "finished training descriptor of dimension %d" %(d)

/home/kevink/code/dense_correspondence/training/training.pyc in run(self, loss_current_iteration, use_pretrained)
    242         if not use_pretrained:
    243             # create new network and optimizer
--> 244             self._dcn = self.build_network()
    245             self._optimizer = self._construct_optimizer(self._dcn.parameters())
    246         else:

/home/kevink/code/dense_correspondence/training/training.pyc in build_network(self)
    129 
    130         return DenseCorrespondenceNetwork.from_config(self._config['dense_correspondence_network'],
--> 131                                                       load_stored_params=False)
    132 
    133     def _construct_optimizer(self, parameters):

/home/kevink/code/dense_correspondence/network/dense_correspondence_network.pyc in from_config(config, load_stored_params, model_param_file)
    412             config["backbone"]["resnet_name"] = "Resnet34_8s"
    413 
--> 414         fcn = DenseCorrespondenceNetwork.get_fcn(config)
    415 
    416         if 'normalize' in config:

/home/kevink/code/dense_correspondence/network/dense_correspondence_network.pyc in get_fcn(config)
    373         if config["backbone"]["model_class"] == "Resnet":
    374             resnet_model = config["backbone"]["resnet_name"]
--> 375             fcn = getattr(resnet_dilated, resnet_model)(num_classes=config['descriptor_dimension'])
    376 
    377         elif config["backbone"]["model_class"] == "Unet":

/home/kevink/code/external/pytorch-segmentation-detection/pytorch_segmentation_detection/models/resnet_dilated.py in __init__(self, num_classes)
    245                                        pretrained=True,
    246                                        output_stride=8,
--> 247                                        remove_avg_pool_layer=True)
    248 
    249         # Randomly initialize the 1x1 Conv scoring layer

/usr/local/lib/python2.7/dist-packages/torchvision/models/resnet.pyc in resnet34(pretrained, progress, **kwargs)
    235     """
    236     return _resnet('resnet34', BasicBlock, [3, 4, 6, 3], pretrained, progress,
--> 237                    **kwargs)
    238 
    239 

/usr/local/lib/python2.7/dist-packages/torchvision/models/resnet.pyc in _resnet(arch, block, layers, pretrained, progress, **kwargs)
    208 
    209 def _resnet(arch, block, layers, pretrained, progress, **kwargs):
--> 210     model = ResNet(block, layers, **kwargs)
    211     if pretrained:
    212         state_dict = load_state_dict_from_url(model_urls[arch],

TypeError: __init__() got an unexpected keyword argument 'fully_conv'

kevinkengne, 2019-07-01:
I figured out what the problem was. Thank you

closed 2019-07-01 by kevinkengne

chrisole, 2020-03-18:
Hi bro,
I got the same error. Could u plz tell me how to solve it?