RobotLocomotion/pytorch-dense-correspondence #206: library path issue in Step 6: Run a simple dataset loader test

ozzie00, 2019-10-24:
when we run pytorch dense correspondence code, we encounter the following issue:

---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-7bbb46fec391> in <module>()
      1 import dense_correspondence_manipulation.utils.utils as utils
      2 utils.add_dense_correspondence_to_python_path()
----> 3 from dense_correspondence.training.training import *
      4 import sys
      5 import logging

/home/robot/code/dense_correspondence/training/training.py in <module>()
     28 import pytorch_segmentation_detection.models.fcn as fcns
     29 import pytorch_segmentation_detection.models.resnet_dilated as resnet_dilated
---> 30 from pytorch_segmentation_detection.transforms import (ComposeJoint,
     31                                                        RandomHorizontalFlipJoint,
     32                                                        RandomScaleJoint,

/home/robot/code/external/pytorch-segmentation-detection/pytorch_segmentation_detection/transforms.py in <module>()
      5 from PIL import Image, ImageOps
      6 
----> 7 import torchvision.transforms.functional as F
      8 
      9 

ImportError: No module named functional

we have checked in the terminal by using the same code, obviously it is right library path. it is very strange why it is not right in Jupyter notebook.

manuelli, 2019-10-24:
Hi,

What notebook were you running when you encountered this error? I have just re-run the `training_tutorial.ipynb` on my local machine and confirmed it is working. I think the error you are seeing is because we use a custom fork of torchvision, not the standard pip version. It is important to follow the steps in the tutorial to ensure that this module gets loaded correctly.

As an example showing that `import torchvision.transforms.functional as F` works try the following in a terminal inside your docker container.

```
manuelli@paladin-44:~/code$ use_pytorch_dense_correspondence 
using pdc
manuelli@paladin-44:~/code$ python
Python 2.7.12 (default, Nov 12 2018, 14:36:49) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import dense_correspondence_manipulation.utils.utils as utils
>>> utils.add_dense_correspondence_to_python_path()
>>> import torchvision
>>> print torchvision.__file__
/home/manuelli/code/external/pytorch-segmentation-detection/vision/torchvision/__init__.pyc
>>> import torchvision.transforms.functional as F
>>> print torchvision.transforms.functions.__file__
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'functions'
>>> print torchvision.transforms.functional.__file__
/home/manuelli/code/external/pytorch-segmentation-detection/vision/torchvision/transforms/functional.pyc

```

 Note that `torchvision` points the the submodule of our repo located at `external/pytorch-segmentation-detection/vision/torchvision`. If instead you see `torchvison.__file__` pointing to a system installed copy namely

```
>>> print torchvision.__file__
/usr/local/lib/python2.7/dist-packages/torchvision/__init__.pyc
>>> 
```

you have a problem. My guess is that you have this error because you didn't clone down the submodules of our repo. This should have been completed in Step 3 of the [tutorial](https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/master/doc/tutorial_getting_started.md). Hopefully this resolves your issue

ozzie00, 2019-11-04:
Hi, Manuelli, 

      we have already cloned the submodules before we had submitted this issue. we will follow your steps to reproduce your resule, thanks.