RobotLocomotion/pytorch-dense-correspondence #208: Why isn't the uv's position an integer?

ierosodin, 2019-11-19:
Hi,

When I want to use your dataset and script to evaluate Keypoint between different scenes, I found that the u, v data have floating point.

Example in evaluation_labeled_data/shoe_green_nike.yaml
----------------------------------------
- image_a:
    image_idx: 0
    pixels:
    - u: 382.0
      v: 197.5
    - u: 341.0
      v: 210.0
    - u: 295.5
      v: 219.5
    - u: 240.0
      v: 246.5
    - u: 273.5
      v: 263.0
    - u: 254.5
      v: 225.5
    scene_name: 2018-05-15-00-04-06
  image_b:
    image_idx: 0
    pixels:
    - u: 192.5
      v: 323.5
    - u: 224.0
      v: 285.0
    - u: 258.0
      v: 222.0
    - u: 294.0
      v: 182.0
    - u: 259.5
      v: 191.5
    - u: 294.5
      v: 208.0
    scene_name: 2018-05-15-00-08-46
----------------------------------------

I'm wondering whether if I sound "round off"?

manuelli, 2019-11-19:
Hi,

Sorry about that we can look into it. Yes it should be fine to cast it to an int. The floating points appear since during labeling we sometimes zoomed in on the image

ierosodin, 2019-11-19:
I see!
Thanks for your quick reply.

Sincerely,
Tony

closed 2019-11-19 by ierosodin