RobotLocomotion/pytorch-dense-correspondence #212: Change Detection Pipeline Not Working

kudo1026, 2020-06-09:
Hi,

I tried to follow the instructions in "change_detection.md" to run the change detection pipeline. However, the environment set up is not working. The `use_director` command only does `PATH=$PATH:~/director/bin`, but apparently this is not enough. When I'm trying to run "run_change_detection_pipeline.py", there are many failed attempts to import director and PythonQt, etc. 

Could you please help fix this pipeline as we are trying to use your work on some other data. 

Thanks!

manuelli, 2020-06-09:
Hey Kudo,

Sorry you ran into a problem. It's an easy fix I believe. That pipeline for actually generating the dataset from a reconstruction is pretty involved and specific to our setup, thus we never really intended for that part of the code base to be used externally. What's in the tutorial is really what we try to support well.

With regards to your specific problem I just ran it on my machine and it's working fine. Are you perhaps running 

```
python run_change_detection_pipeline.py
```

instead of just 

```
run_change_detection_pipeline.py
```

In order for the `director` and `PythonQt` imports to work you need to run the scripts using `directorPython` instead of `python`. To give you a simple example of this look at the `run_change_detection.py` script. Notice that at the top it has [`#!/usr/bin/env directorPython
`](https://github.com/RobotLocomotion/pytorch-dense-correspondence/blob/master/modules/dense_correspondence_manipulation/scripts/run_change_detection.py#L1). Thus on my machine

```
run_change_detection.py --data_dir /home/manuelli/data/scratch/2018-04-10-16-05-17/processed
```

works just fine while

```
python run_change_detection.py --data_dir /home/manuelli/data/scratch/2018-04-10-16-05-17/processed
```

errors out with

```
Traceback (most recent call last):
  File "run_change_detection.py", line 6, in <module>
    import dense_correspondence_manipulation.change_detection.change_detection as change_detection
  File "/home/manuelli/code/modules/dense_correspondence_manipulation/change_detection/change_detection.py", line 6, in <module>
    from director import imageview
ImportError: No module named director

```

Hopefully that should fix your problem.

kudo1026, 2020-06-10:
Thanks for the speedy response! 
You are exactly right, I was running
``python run_change_detection_pipeline.py``
instead of
``run_change_detection_pipeline.py``
I'm sorry I just thought it was a typo in the doc and overlooked it.
And it works alright now, thanks for support. Love it!

closed 2020-06-10 by kudo1026

peteflorence, 2020-06-10:
Awesome debugging @manuelli !