RobotLocomotion/pytorch-dense-correspondence #216: Weird pip version mismatch when running docker_build.py

WangYixuan12, 2021-01-29:
Hi guys! I am going through the tutorial. However, it reports a weird error when running docker_build.py as below:
```
+ pip install jupyter opencv-python plyfile pandas tensorflow future typing
Collecting jupyter
  Downloading https://files.pythonhosted.org/packages/83/df/0f5dd132200728a86190397e1ea87cd76244e42d39ec5e88efd25b2abd7e/jupyter-1.0.0-py2.py3-none-any.whl
Collecting opencv-python
  Downloading https://files.pythonhosted.org/packages/a1/d6/8422797e35f8814b1d9842530566a949d9b5850a466321a6c1d5a99055ee/opencv-python-4.3.0.38.tar.gz (88.0MB)
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-B7IEGQ/opencv-python/setup.py", line 9, in <module>
        import skbuild
    ImportError: No module named skbuild
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-B7IEGQ/opencv-python/
You are using pip version 9.0.3, however version 21.0 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```
I am new to docker and I don't quite understand why this happened. pip on my local machine is 20.3.4. My machine is Ubuntu 18.04 and my output from `nvidia-docker run --rm nvdia/cuda:11.0-base nvidia-smi` is shown below if it helps.
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 450.102.04   Driver Version: 450.102.04   CUDA Version: 11.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  GeForce RTX 208...  Off  | 00000000:01:00.0  On |                  N/A |
| 41%   33C    P8    11W / 260W |    367MiB / 11016MiB |      2%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+
```

WangYixuan12, 2021-01-29:
By the way, I have run `pip install --upgrade pip` as suggested but it tells me
```
Defaulting to user installation because normal site-packages is not writeable
Requirement already up-to-date: pip in /home/deformtrack/.local/lib/python2.7/site-packages (20.3.4)
```
I think it just means it is not upgradable.

WangYixuan12, 2021-01-31:
I think `pytorch-dense-correspondence.dockerfile` and `install_*.sh` are a bit outdated, e.g. version conflicts of newly released dependencies. So I modified them a bit and push them on my own fork of dense correspondence. It builds the docker image successfully, but I did NOT run it yet. But feel free to refer to [it](https://github.com/WangYixuan12/pytorch-dense-correspondence)!

closed 2021-01-31 by WangYixuan12

spirosperos, 2021-07-20:
I have the same issue.
`File "/tmp/pip-build-B7IEGQ/opencv-python/setup.py", line 9, in <module>
        import skbuild
    ImportError: No module named skbuild`

spirosperos, 2021-08-10:
If anyone else still has this problem I have opened a [pull request ](https://github.com/RobotLocomotion/pytorch-dense-correspondence/pull/217/commits/c2f0fc42f1a893faa4ae2669a8241018d200fcad) which has the updated package versions that make it possible to finish the whole tutorial in August 2021.